lastralab/ABISSMAL #472: [NEW] R coding and video tutorials

gsvidaurre, 2023-12-27:
Make short R coding tutorials (English and Spanish) along with videos if possible. Also consider adding videos of basic ABISSMAL setup

gsvidaurre, 2024-03-21:
From #473:

All R code that calls the functions need to be modified to specify the new repository name. Also, I found a bug in preprocess_detections while working on the tutorials with simulated data. It appears that the detections aren't being filtered out correctly. I need to find out why and also check why the unit tests didn't catch this bug before

The only difference I can find between the simulated datasets for the tutorials and those for the unit testing are that the former end on an odd numbered timestamp and the latter end on an even number. So the logic inside preprocess_detections() works for series of timestamps that end in even numbers but not for those that end with odd numbers. Need to fix the logic and add more simulated data that end in odd numbered timestamps to the unit tests

gsvidaurre, 2024-03-21:
I checked the code inside preprocess_detections() and checked the output from data processing in the vignettes, and everything looks as expected. I was imagining the bug in data processing. Also, the unit tests for preprocess_detections() are generally written with an odd-numbered series of timestamps. I added another unit test with an even-numbered series of timestamps, and that test passed as well.

gsvidaurre, 2024-03-21:
Just updated the repo name to be capitalized in all unit test files

gsvidaurre, 2024-03-25:
For Spanish translations I plan to follow the Carpentries translation guidelines: https://github.com/Carpentries-ES/board/blob/master/Convenciones_Traduccion.md, including their guidelines for technical terms. See the Carpentries GitHub page for Spanish translations they have done: https://github.com/Carpentries-ES. Also see Glosario, their multi-language collection of technical terms: https://carpentries.org/blog/2020/07/announcing-glosario/ that is available as a package for inserting glossary links for learning more about specific data science terms

lastralab, 2024-04-26:
Merged Vignettes and Tutorials

closed 2024-04-26 by lastralab